MikePopoloski/pyslang #59: How to parsing nested relationships  using pyslang？

fzwwj95, 2022-11-19:
Hi，How to parsing nested relationships  using pyslang？
Suppose my code looks like this：
`    sequence s1;
        $rose(ctrl);
    endsequence
    sequence ctrl_rose;
        @(posedge clk) s1;
    endsequence
    ...
    ...
    property F1;
        ctrl_rose |=> result;
    endproperty
    a1: assert property(F1);`
How can I parsing “ctrl_rose” and get the source code “@(posedge clk) $rose(ctrl) ”？

MikePopoloski, 2022-11-20:
Given:
```sv
module m;
    wire ctrl,clk,result;
    sequence s1;
        $rose(ctrl);
    endsequence

    sequence ctrl_rose;
        @(posedge clk) s1;
    endsequence

    property F1;
        ctrl_rose |=> result;
    endproperty
    a1: assert property(F1);
endmodule
```

The following script:
```py
import pyslang
from pyslang import *

c = Compilation()
c.addSyntaxTree(SyntaxTree.fromFile("path/to/file.sv"))

# Get the concurrent assertion statement
a1 = c.getRoot().topInstances[0].body[7].body.body

# The reference to F1 property
print(a1.syntax)

# The body of F1 property
f1 = a1.propertySpec.expr.body
print(f1.syntax)

# The body of the ctrl_rose sequence
ctrl_rose = f1.left.expr.body
print(ctrl_rose.clocking.syntax)

# The body of the s1 sequence
s1 = ctrl_rose.expr.expr.body
print(s1.syntax)
```

Will print:
```
    a1: assert property(F1);

        ctrl_rose |=> result

        @(posedge clk)

        $rose(ctrl)
```

fzwwj95, 2022-11-20:
Thanks for the example above. Can pyslang parse out “@(posedge clk) $rose(ctrl) ” from  “ctrl_rose” automatically？When I met a large design ，I don't know how many layers of nested. At this time, only  can I use Python to judge the Kind of each layer and parse  the next layer?

MikePopoloski, 2022-11-20:
Yes, all of the information should be there. It's not the easiest thing to figure out from the Python side, it's helpful to look at the C++ types, but you should be able to write a script that can do it.

fzwwj95, 2022-11-20:
> Yes, all of the information should be there. It's not the easiest thing to figure out from the Python side, it's helpful to look at the C++ types, but you should be able to write a script that can do it.
Thank you for the answers above.
I am so sorry ，I did not find the function or class to print the AST or turn the AST to json through the API. Could you give me a hint，millions of thanks

MikePopoloski, 2022-11-20:
There is an ASTSerializer class in C++ that has not yet been exposed to the Python side, but that could be done so without too much difficulty. It also shouldn't be that hard to write a Python function that just walks the AST via Python's built-in introspection tools and serialize everything that way; probably there is some Python library that will just do it automatically but I haven't looked.

closed 2022-11-20 by fzwwj95