MikePopoloski/pyslang #65: Is it possible to use Rewriter with python binding

jahagirdar, 2022-11-26:
I am looking at using pyslang to solve the common problem of `ip's from different vendors have files with same name` so
`Add an prefix to all modules, their instances and files`
Is this possible using pyslang or should I write a c code for this?

MikePopoloski, 2022-11-26:
The Rewriter is not currently exposed to Python but it could be. If you don't want to wait for that or create a PR to do so then you'd have to use the C++ API.

MikePopoloski, 2024-06-06:
Following up here since it's been a while -- the C++ API is capable of doing what is requested. Almost none of the syntax and AST creation methods are exposed to Python right now, so it would be a decent amount of work to get that all exposed in order to let Python do syntax rewriting / mutation. I'd be happy to accept a PR if someone wants to add it but it's not something I can spend time on in the near future.

Badhi, 2024-09-06:
I m wondering if you really need a rewriter for this. Can't you use pyslang to find the locations where prefixes need to be added and simply do a text replacement on the source text buffer