MikePopoloski/pyslang #113: Access to preprocessor tokens/directives from PySlang API

svenka3, 2023-04-27:
Hello,
  Is there a way to get access to preprocessor directives via PY API in PySlang. For instance,

```verilog
`define WIDTH 8
`define MY_FUNC my_function();
``` 
I believe the Preprocessor in the C++ API handles this, what about the PY interface? Do we have access to this in PY? If so, any pointers would be appreciated. 

Thanks
Srini

MikePopoloski, 2023-04-29:
Yes, they are accessible. It depends on what you're doing, but if you have a SyntaxNode created by the parser, you can look at the tokens in that syntax node, look at the trivia inside those tokens, and find the preprocessor definitions that immediately preceded them.

svenka3, 2023-05-03:
Thanks for the response. I don't see how to get to the trivia on the below (trivial) code:

```verilog
  `define WIDTH 8
  `define FIRST_MACRO sig_one
```

Can you get me started, please?

MikePopoloski, 2023-05-05:
You can read a high level overview of syntax nodes, tokens, and trivia here: https://sv-lang.com/parsing.html#syntax-nodes

Some example Python code for your given snippet:
```
>>> import pyslang
>>> tree = pyslang.SyntaxTree.fromText("`define WIDTH 8\n`define FIRST_MACRO sig_one")
>>> tree.root
SyntaxNode(SyntaxKind.CompilationUnit)
>>> tree.root.endOfFile.trivia
[Trivia(TriviaKind.Directive), Trivia(TriviaKind.Directive)]
>>> for t in tree.root.endOfFile.trivia:
...   print(t)
...   print(t.syntax())
...
Trivia(TriviaKind.Directive)
`define WIDTH 8
Trivia(TriviaKind.Directive)
`define FIRST_MACRO sig_one
```

Nalaka1693, 2023-12-08:
Hi Thank you for the amazing library. 
Is there a way to access macros in the same text with a module?
```
>>> import pyslang
>>> tree = pyslang.SyntaxTree.fromText("`define WIDE 11\n\nmodule mod_name (\noutput [`WIDE:0] inst_data,\n    input [`WIDE:0] instr_0\n    );\n\n    assign instr_0 = inst_data;\nendmodule\n")
>>> tree.root
SyntaxNode(SyntaxKind.ModuleDeclaration)
>>> tree.root.endOfFile
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'pyslang.ModuleDeclarationSyntax' object has no attribute 'endOfFile'
>>> print(tree.root.parent)
None
```

MikePopoloski, 2023-12-10:
```
>>> for t in tree.root.header.moduleKeyword.trivia:
...   if t.kind == pyslang.TriviaKind.Directive:
...     print(t.syntax())
...
`define WIDE 11
```