MikePopoloski/pyslang #115: Can not build

17Reset, 2023-05-01:
Why can't I build pyslang successfully?

MikePopoloski, 2023-05-01:
I don't know, why can't you?

17Reset, 2023-05-02:
> I don't know, why can't you?

Tips like this：Building wheel for pyslang (pyproject.toml) ... error
error: subprocess-exited-with-error

× Building wheel for pyslang (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [138 lines of output]
fatal: not a git repository (or any of the parent directories): .git
WARNING: '' not a valid package name; please use only .-separated package names in setup.py

MikePopoloski, 2023-05-02:
You need to give more information if you want help. What commands exactly did you run? What is the full output of running those commands?

17Reset, 2023-05-02:
> You need to give more information if you want help. What commands exactly did you run? What is the full output of running those commands?
Sorry, the problem I encountered is as follows:
D:\App\Python\Scripts>pip install pyslang
Collecting pyslang
  Using cached pyslang-3.0.0.tar.gz (945 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pyslang
  Building wheel for pyslang (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for pyslang (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [129 lines of output]
      fatal: not a git repository (or any of the parent directories): .git
      WARNING: '' not a valid package name; please use only .-separated package names in setup.py
      C:\Users\AppData\Local\Temp\pip-build-env-ilxzfons\overlay\Lib\site-packages\setuptools\config\setupcfg.py:293: _DeprecatedConfig: Deprecated config in `setup.cfg`
      !!

              ********************************************************************************
              The license_file parameter is deprecated, use license_files instead.

              By 2023-Oct-30, you need to update your project and remove deprecated calls
              or your builds will no longer be supported.

              See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
              ********************************************************************************

      !!
        parsed = self.parsers.get(option_name, lambda x: x)(value)

      --------------------------------------------------------------------------------
      -- Trying 'Ninja (Visual Studio 17 2022 x64 v143)' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      Not searching for unused variables given on the command line.
      -- The C compiler identification is MSVC 19.35.32217.1
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: D:/App/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx86/x64/cl.exe - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- The CXX compiler identification is MSVC 19.35.32217.1
      CMake Warning (dev) at C:/Users/zhujiang/AppData/Local/Temp/pip-build-env-ilxzfons/overlay/Lib/site-packages/cmake/data/share/cmake-3.26/Modules/CMakeDetermineCXXCompiler.cmake:168 (if):
        Policy CMP0054 is not set: Only interpret if() arguments as variables or
        keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
        details.  Use the cmake_policy command to set the policy and suppress this
        warning.

        Quoted variables like "MSVC" will no longer be dereferenced when the policy
        is set to NEW.  Since the policy is not set the OLD behavior will be used.
      Call Stack (most recent call first):
        CMakeLists.txt:4 (ENABLE_LANGUAGE)
      This warning is for project developers.  Use -Wno-dev to suppress it.

      CMake Warning (dev) at C:/Users/zhujiang/AppData/Local/Temp/pip-build-env-ilxzfons/overlay/Lib/site-packages/cmake/data/share/cmake-3.26/Modules/CMakeDetermineCXXCompiler.cmake:189 (elseif):
        Policy CMP0054 is not set: Only interpret if() arguments as variables or
        keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
        details.  Use the cmake_policy command to set the policy and suppress this
        warning.

        Quoted variables like "MSVC" will no longer be dereferenced when the policy
        is set to NEW.  Since the policy is not set the OLD behavior will be used.
      Call Stack (most recent call first):
[... 123 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^
  File "D:\App\Python\Lib\concurrent\futures\_base.py", line 401, in __get_result
    raise self._exception
  File "D:\App\Python\Lib\site-packages\pip\_vendor\tenacity\__init__.py", line 384, in __call__
    result = fn(*args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^
  File "D:\App\Python\Lib\site-packages\pip\_internal\utils\misc.py", line 130, in rmtree
    shutil.rmtree(dir, ignore_errors=ignore_errors, onerror=rmtree_errorhandler)
  File "D:\App\Python\Lib\shutil.py", line 759, in rmtree
    return _rmtree_unsafe(path, onerror)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\App\Python\Lib\shutil.py", line 617, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "D:\App\Python\Lib\shutil.py", line 617, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "D:\App\Python\Lib\shutil.py", line 617, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  [Previous line repeated 6 more times]
  File "D:\App\Python\Lib\shutil.py", line 626, in _rmtree_unsafe
    onerror(os.rmdir, path, sys.exc_info())
  File "D:\App\Python\Lib\shutil.py", line 624, in _rmtree_unsafe
    os.rmdir(path)
OSError: [WinError 145] 目录不是空的。: 'C:\\Users\\\AppData\\Local\\Temp\\pip-install-k9d462z2\\pyslang_d82763474bf744e5a58005a04ebefdbd\\_skbuild\\win-amd64-3.11\\cmake-build\\_deps\\unordered_dense-subbuild\\unordered_dense-populate-prefix\\src\\unordered_dense-populate-stamp'

D:\App\Python\Scripts>

17Reset, 2023-05-02:
> These commands all succeed for me when I run them. Do you have a special setup for git? It seems like there's some problem with your environment or your git installation that is preventing it from working correctly.

When I run: git clone https://github.com/MikePopoloski/pyslang.git, no problem at all.
But run: git submodule update - init -- recursive, don't seem to anything from https://github.com/MikePopoloski/slang.git.

17Reset, 2023-05-02:
I still hope to know why those errors occur when using pip insatall pyslang. Thank you very much indeed

17Reset, 2023-05-02:
> It seems like the same problem, some git operation fails during the build of the pyslang wheel. Your environment must unique and/or messed up in some way. Typically pip install doesn't require you to build a binary at all because the package has binaries for common systems already included, so you must be using an uncommon system.

What is an uncommon system?

MikePopoloski, 2023-05-02:
You can see the list of prebuilt binaries here: https://pypi.org/project/pyslang/#files

17Reset, 2023-05-02:
> 您可以在此处查看预构建二进制文件的列表：https://pypi.org/project/pyslang/#files

Is python3.11 not supported?

17Reset, 2023-05-02:
I see what's going on. Is it the wrong version of python, pyslang doesn't support python311

MikePopoloski, 2023-05-02:
I haven't tried to build it for py3.11 yet but I would be surprised if that was the cause of the build failure, it seems unrelated. Not having py3.11 prebuilt binaries in PyPi though is why you are having to build locally in the first place.