MikePopoloski/pyslang #118: TB Lint on top of pyslang - pyslint?

svenka3, 2023-05-04:
Hi,
  I've been playing with pyslang API and I must say I like it quite a bit, kudos to the developers. I now have a set of style rules/lint checks for SV-TB based on the PY bindings available so far. It is far from perfect as I am reading the code to understand the object-model/PY API per-se. But, I do have a good start with a bunch of naming convention checks etc. 

How do you @MikePopoloski envision this to develop - if any thoughts? I see a similar dev in dalance/svlint based on Rust, but I prefer PY for now. Should I just build a separate repo say "pyslint/pysvlint" or something or do you see that name too close to comfort? Any other thoughts/comments would be great. Naturally, I will seek more help here to add more rules, especially around the areas that I don't fully understand on the object-model.

Thanks

MikePopoloski, 2023-05-05:
Sure, I'm happy for you to build out linting capabilities in a separate repo. The proposed names are fine with me.

I'm happy to answer questions. Unfortunately slang is generally lacking in documentation, and what I do have is written with C++ in mind, not Python, so there is some exploration involved with figuring out how to use the Python API. If you wanted to help contribute docs that would also be welcome.

svenka3, 2023-05-10:
Thanks, first few rules are up via: https://github.com/svenka3/pyslint 

I will close this issue for now.

closed 2023-05-10 by svenka3