MikePopoloski/pyslang #138: static keyword eaten up during parsing?

svenka3, 2023-06-08:
In SV code below:

```verilog
class c;
  extern function void void_f ();
endclass : c
static function void  c::void_f();
endfunction : void_f
```

The above code is not 100% LRM, but works with atleast 2 simulators. With pyslang code below, I see that "static" keyword seems to be "eaten" up and I am unable to "detect" such wrong usage. Any clues please?

```python
import pyslang
tree = pyslang.SyntaxTree.fromFile('test.sv')

for s in (tree.root.members):
  print (s)
```
I get log as below:
```log
class c;
  extern function void void_f ();
endclass : c
 function void  c::void_f();
endfunction : void_f
```

MikePopoloski, 2023-06-08:
The invalid token is skipped and prepended as trivia. By default the printer methods don't include skipped tokens; you can use the SyntaxPrinter class and set the options manually in order to see them. For example, using the printFile helper method:
```
pyslang.SyntaxPrinter.printFile(tree)
```

svenka3, 2023-06-08:
Thanks, can you please show how to access this via trivia in this example? Print is just to demo the issue I used here, in actual lint code I need to check for that static keyword as part of qualifiers/attributes/declaration etc. so maybe I can use trivia in this case? I am unable to access the trivia, so an example would be very useful 

Thanks

MikePopoloski, 2023-06-09:
```
tree.root.members[1].getFirstToken().trivia[0].getSkippedTokens()
[Token(TokenKind.StaticKeyword)]
```

svenka3, 2023-06-09:
Great, thanks, it works fine now.

closed 2023-06-09 by svenka3