MikePopoloski/pyslang #140: Using UVM parsing in pyslang fromFiles - need help

svenka3, 2023-06-09:
Hi,
  How do I pass UVM and a test along with it to pyslang? I tried below:

```python
af_uvm_opts = pyslang.Bag
af_src_mgr = pyslang.SourceManager

#af_uvm_opts.compilationOptions('+incdir=../uvm_tests/uvm-1.1d/src')

#tree = pyslang.SyntaxTree.fromFile(inp_test_name)
tree = pyslang.SyntaxTree.fromFiles(paths="../uvm_tests/uvm-1.1d/src/uvm_pkg.sv", sourceManager=af_src_mgr, options=af_uvm_opts)
```
It does not work, am sure it is wrong - but can't figure out how to fix, can you please help? Or show a working example of UVM parsing in pyslang.

Thanks

(below is the log)
```log
  File "../py_src/pyslint.py", line 691, in <module>
    tree = pyslang.SyntaxTree.fromFiles(paths="../uvm_tests/uvm-1.1d/src/uvm_pkg.sv", sourceManager=af_src_mgr, options=af_uvm_opts)
TypeError: fromFiles(): incompatible function arguments. The following argument types are supported:
    1. (paths: span[str]) -> pyslang.SyntaxTree
    2. (paths: span[str], sourceManager: pyslang.SourceManager, options: pyslang.Bag = <pyslang.Bag object at 0x7f6bb27d1b30>) -> pyslang.SyntaxTree

```

MikePopoloski, 2023-06-09:
`paths` takes a span, so you want to pass something like an array of paths to the files you want to parse.

svenka3, 2023-06-09:
Thanks, the following gets better, but not yet fully done.
```python
af_uvm_files = ["../uvm_tests/uvm-1.1d/src/uvm_pkg.sv", inp_test_name]

tree = pyslang.SyntaxTree.fromFiles(paths=af_uvm_files, sourceManager=af_src_mgr, options=af_uvm_opts)

```
I believe this is because we need to add +incdir to compile options (When I run plain slang it throws clear errors, pyslang SyntaxTree API does not report any, any way to get all errors to STDOUT please).

I tried adding it via Bag as below:
```python
af_uvm_opts = pyslang.Bag()
af_src_mgr = pyslang.SourceManager()

af_uvm_opts.compilationOptions('+incdir=../uvm_tests/uvm-1.1d/src')
```
But I get error as compilationOptions is None. 

Thanks

MikePopoloski, 2023-06-09:
Include directories are set on the SourceManager, with `addUserDirectory`.

Diagnostics are in the SyntaxTree's `diagnostics` member.

svenka3, 2023-06-09:
Cool, it works now, thanks!

closed 2023-06-09 by svenka3