MikePopoloski/pyslang #141: UVM parsing - retain preprocessor directives, howto?

svenka3, 2023-06-09:
```verilog
import uvm_pkg::*;
`include "uvm_macros.svh"

module m;
  initial begin : test
    `uvm_info ("TEST", "Hello World", UVM_LOW)
  end
endmodule : m
```

When pyslang parses this code and I print the code from SyntaxTree I get the expanded version than the origin `uvm_info. How do I access the unprocessed version to run a simple check such as `uvm_info should not have a semicolon at the end?

Sample log:
```log

   begin
     if (uvm_report_enabled(UVM_LOW,UVM_INFO,"TEST"))
       uvm_report_info ("TEST", "Hello World", UVM_LOW, "../uvm_tests/test_uvm_hello_p.sv", 6);
   end
```

My wish-log:
```log
 `uvm_info ("TEST", "Hello World", UVM_LOW)
```
Thanks

MikePopoloski, 2023-06-12:
When printing, you can use the SyntaxPrinter's includeDirectives=true option and includePreprocessed=false option. You can see what they do here: https://github.com/MikePopoloski/slang/blob/master/source/syntax/SyntaxPrinter.cpp

MikePopoloski, 2023-06-24:
Closing this since it appears to be answered.

closed 2023-06-24 by MikePopoloski