MikePopoloski/pyslang #145: Git version crashes when used with other libraries

EmanuelFtng, 2023-06-14:
Since the possibility to use the visitor is not yet available via pip (pyslang version 3.0.0), I have to use the git repository. With the cloned version the visitor works great. However, there are now problems with other libraries. As soon as I use the cloned version of pyslang and at the same time for example [streamlit](https://github.com/streamlit) installed via pip I get a `Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)` error.

To reproduce the error it is enough to run a file with the following two lines:
import pyslang
import streamlit

Did I miss something or is there a solution to the problem?

MikePopoloski, 2023-06-24:
Unfortunately this is not enough information to reproduce the problem. I install pyslang via local checkout, installed streamlit, and ran your two lines without trouble. Do you have better steps for reproducing?

EmanuelFtng, 2023-06-25:
Thanks for your reply. What system are you working on and what version of Python are you using? I just tried 3 different combinations and it didn't work with any of them.

For all of them I proceeded as described in the README:
1) git clone https://github.com/MikePopoloski/pyslang.git
2) cd pyslang
3) git submodule update --init --recursive
4) pip install .
5) pip install streamlit
6) python
6.1) >>> import pyslang
6.2) >>> import streamlit
-> segmentation fault (core dumped) python

**Running imports did not work for:**
Arch with Python 3.11.3
Debian 11 with Python 3.9.2
Ubuntu 22.04 with Python 3.10.6

MikePopoloski, 2023-06-25:
Ok, I have reproduced now with Ubuntu 22.04 and Python 3.10. I'll investigate. It seems like the second import isn't actually needed, trying to do almost anything except using pyslang results in a segfault.

MikePopoloski, 2023-06-25:
I think the problem was the use of mimalloc in the python builds. I committed e7d434d3ded712b804bb767a471f19bf7b208b6f upstream -- please try it out and see if it resolves your issue.

EmanuelFtng, 2023-06-26:
Everything is working again. Thanks a lot :)