MikePopoloski/pyslang #170: Command file loading with options

Cyberwizzard, 2023-07-28:
I am attempting to load a file list which I think slang calls "command files". From what I gather, the pyslang.Driver class is the correct loader for this as it handles the loading of files and include paths automatically.

From some experiments, I gather that the procedure to use this class and get a compiled and elaborated tree is:
```
d: pyslang.Driver = pyslang.Driver()
# For some reason I need to call addStandardArgs() or we get errors later on?
d.addStandardArgs()
assert d.processCommandFile('/home/cyberwizzard/ip/SomeIP/rtl/SomeIP.fl', True) == True
# After loading all files into buffers, parse them now into syntax trees
print(d.parseAllSources())
# Create a compilation for all the loaded syntax trees
comp: pyslang.Compilation = d.createCompilation()
# Perform elaboration by requesting the root of the compilation
root: Symbol = comp.getRoot()
diags = comp.getAllDiagnostics()
if diags:
    report = pyslang.DiagnosticEngine.reportAll(comp.sourceManager, diags)
    print(report)
    sys.exit(1)

print("Compile and elaborate OK")
```

In my case, I need to set a few defines and I want to ignore other top-level modules aside from one, which when using pure slang I can just do by specifying the top level and adding the defines to the command line.

From Python, I see that I can get an option Bag object:
```
bag: pyslang.Bag = d.createOptionBag()
bag.preprocessorOptions.predefines.append('USE_WORKAROUND')
bag.compilationOptions.topModules.add('MyRealTop')
```
But there seems to be no way to return the options back to the driver as addParseOptions() is a private function and the changes above do not seem to influence the compile at all, so ```bag``` is not a reference to a shared object.

My other idea was to set the options in the Compile instance, but pre-processor directives are already applied, so that is not the solution either. Attempting to set the top-level via the compilation instance also did not work as it yields multiple top-levels, but not the one I specified (I set this before calling `getRoot()`).
```
comp.options.topModules.add('MyRealTop')
```

So how does one from within Python using pyslang:
1. Define the top-level I want to elaborate
2. Set pre-processor defines
3. Suppress warnings (either selectively, or all)

If there are some examples somewhere I must apologize as I have not found them.

Cyberwizzard, 2023-07-28:
Also perhaps good to mention that MyRealTop is a module which has been integrated in some other modules which are also in the file list, and which I can not filter out easily - so I just want to compile MyRealTop and its sub-modules.

MikePopoloski, 2023-07-28:
- They're called "command files" (not a term I invented) because you can put any command line arguments in them, not just file lists. 
- If you don't call addStandardArgs then no command line arguments are registered and so you get errors about unknown arguments. They aren't added automatically because the idea is that you might use the Driver functionality but with your own set of command line arguments defined.
- Probably the easiest way to get what you want is to call parseCommandLine() with a string containing the additional options you want to set.

Cyberwizzard, 2023-07-28:
I have tried this:
```
d: pyslang.Driver = pyslang.Driver()
d.addStandardArgs()
d.processCommandFile('/pathToSomeIP/SomeIP.fl', True)
# When not calling parseAllSources first, there is an error about SomeIP not being an existing path
d.parseAllSources()
# This does nothing
d.parseCommandLine("--top SomeIP -DSOME_DEFINE -Wno-width-trunc")
comp: pyslang.Compilation = d.createCompilation()
```

But the defines are not applied and the top I requested is not actually listed in the `topModules`. I also tried adding the command file using `-f` in there, but that gave me different errors.

Additionally, was I correct in my assumption that calling `comp.getRoot()` indeed elaborates the design where possible?

MikePopoloski, 2023-07-28:
You need to parseCommandLine before parseAllSources. I realized you are also missing a processOptions() call. The flow is: addStandardArgs -> parseCommandLine -> processOptions -> parseAllSources. They're broken apart to allow you to customize what happens in between.

Yes, getRoot starts the elaboration process, but it's done lazily as you examine the symbol tree. If you want to force full elaboration, calling getAllDiagnostics is the easiest way to accomplish that.

Cyberwizzard, 2023-07-28:
Thanks for the swift replies :)

I've now refactored my code to do the steps you outline; when I specify a top-level module using `--top SomeIP`, I get:
```
error: no such file or directory: 'SomeIP'
```

When I skip the `--top`, the call to `parseCommandLine` returns True and something seems to get compiled, but not the top I want.

MikePopoloski, 2023-07-28:
Ah, I see what's happening. parseCommandLine assumes the first argument is the name of the program being run, so your `--top` is being eaten up as that. There is an option to not do that but it's not exposed to Python, so as a workaround you can just put some dummy string in there as the first argument.

Cyberwizzard, 2023-07-28:
Ah thanks, that helped a lot!

I am digging in the tree structure to find the elaboration results, to be precise a parameter with a computed default value. From the AST I dumped using `slang` I thought I would find that in the SyntaxNode describing the parameter, but I can only find the initializer which is just the function refence, but not the constant value that comes out of it.

I am looking at: `root.compilationUnits[#].syntax.members[#].header.parameters.declarations[#]`, which is a `ParameterDeclaration`, which has an `initializer` of type `EqualsValueClause`, which in turn has an `expr` of type `InvocationExpression`.

But I do not see any node which contains the value; do I need to evaluate that somehow?

I nearly got all the information I need, just these computed values are missing.

MikePopoloski, 2023-07-28:
The SyntaxNode tree is the *concrete* or *parse* tree. To do name resolution, constant evaluation, etc you need to build the AST / do elaboration, which is what the Compilation object and the symbol tree will give you.

MikePopoloski, 2023-07-28:
To be more clear, shouldn't inspect the syntax member of symbols unless you really want to see what was parsed exactly. You should just look for the ParameterSymbol that you want.

MikePopoloski, 2023-07-29:
I committed a change upstream to add ParseOptions to the parseCommandLine function, which should clean up the issue observed above. I'm going to close this out since it seems all issues have been addressed.

closed 2023-07-29 by MikePopoloski