MikePopoloski/pyslang #172: Unable to get elaborated parameter values from module CompilationUnit

Cyberwizzard, 2023-07-31:
This is a follow-up from the previous issue, where we try to get the elaborated value of a parameter belonging to a module.

Compilation and elaboration is performed as described in issue #170, yielding a `comp` object holding the elaborated set of modules and packages.

This is the code I am trying to run:
```
...
# Code from: https://github.com/MikePopoloski/slang/blob/master/tests/unittests/ast/LookupTests.cpp
# cu is of the type CompilationUnitSymbol, but can also be gotten like this:
# cu = comp.getRoot().compilationUnits[0]

result: pyslang.LookupResult = pyslang.LookupResult()
context: pyslang.ASTContext = pyslang.ASTContext(cu, pyslang.LookupLocation.max)
# Note: the 'None' value in this class works fine in C++ but not in Python3, use the getattr function to work around that
lookupFlag = getattr(pyslang.LookupFlags, 'None')
pyslang.Lookup.name(comp.parseName("ADDR_WIDTH"), context, lookupFlag, result)

# The line below fails as the symbol lookup yielded an error
assert result.found == True
assert result.found.kind == pyslang.SymbolKind.Parameter
```

Inspecting the `result` object shows that `isError` is True and the `result.selectors` value is:
```
TypeError: Unregistered type : slang::SmallVector<std::variant<slang::syntax::ElementSelectSyntax const*, slang::ast::LookupResult::MemberSelector>, 4ul>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
TypeError: Unable to convert function return value to a Python type! The signature was
	(self: pyslang.LookupResult) -> slang::SmallVector<std::variant<slang::syntax::ElementSelectSyntax const*, slang::ast::LookupResult::MemberSelector>, 4ul>

Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>,
<pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic
conversions are optional and require extra headers to be included
when compiling your pybind11 module.
```

Did we convert the C++ unit test incorrectly? Or do we need to do this differently in pyslang?

Cyberwizzard, 2023-07-31:
Porting the unit test performing the lookup to Python works, even though the `result.selectors` field still has the same message.

This was the ported code:
```
from pyslang import *

tree = SyntaxTree.fromText(
"""
package Foo;
    parameter int x = 4;
endpackage

import Foo::x;
"""
)

compilation = Compilation()
compilation.addSyntaxTree(tree)

unit = compilation.getRoot().compilationUnits[0];
lookupFlag = getattr(LookupFlags, 'None')
result = LookupResult()
context = ASTContext(unit, LookupLocation.max)
ident = compilation.parseName("x")
Lookup.name(ident, context, lookupFlag, result)

assert result.wasImported == True
assert result.found != None
assert result.found.kind == SymbolKind.Parameter
assert result.found.value == 4
```

Which means we must be doing something else wrong.

Cyberwizzard, 2023-07-31:
It turns out that compilation.parseName(string) does not do any lookup or checks, it just converts a string RTL path to an object.

The correct method for getting the value of the parameter is by replacing `comp.parseName("ADDR_WIDTH")` with `comp.parseName("MyTopLevelIP.ADDR_WIDTH")`.

This point `result.found` points to an object (instead of `None`) and `isError` is false. The elaborated value is found in `result.found.value`.

@MikePopoloski shall I make a new ticket for the error message in the `results.selectors` field?

Cyberwizzard, 2023-07-31:
And to add one more question since I can not figure out why this does not work: how do I get the same object (`SymbolKind.Parameter`) when inspecting the parameter list? For ports this seems to work, but in the snippet below I get instances of `ParameterSymbolBase` which has 4 fields (`hasDefault`, `isBodyParam`, `isLocalParam` and `isPortParam`) but no way to get to the parameter symbol describing the name, type or value.

Is there a way to look this up?

```
...
result: pyslang.LookupResult = pyslang.LookupResult()
ident = comp.parseName("$unit::MyTopLevel")
pyslang.Lookup.name(ident, context, lookupFlag, result)

assert result.found is not None
assert result.found.isModule == True
assert result.found.kind == pyslang.SymbolKind.Instance

print("Module %s has %i parameters:" % (result.found.name, len(result.found.body.parameters)))
for param in result.found.body.parameters:
    # param is of type ParameterSymbolBase which seems to have no link to the parameter object holding the name, type and value
    print(param)
```

MikePopoloski, 2023-08-02:
I committed the downcaster change for ParameterSymbolBase. I looked into exposing LookupResult::selectors to Python but it's not straightforward and probably not all that useful either so I'm going to skip that for now.

closed 2023-08-02 by MikePopoloski