MikePopoloski/pyslang #184: Code complexity analysis via Pyslang?

svenka3, 2023-08-17:
I looked at extract_comments test here in this repo and am wondering what it takes to do the opposite - to report various stats such as the ones done by SCC: https://github.com/boyter/scc 

1. Active lines of code (total_lines - comments - blank lines)
2. Number of inputs/outputs/nets etc.

Thanks for any pointers.

MikePopoloski, 2023-08-17:
Well counting inputs / outputs etc can be done by traversing the concrete syntax tree (or the abstract syntax tree too) and counting whatever it is you want to count. Counting lines can be done by inspecting trivia members of the tokens produced by the lexer, or by traversing tokens via the concrete syntax tree.

svenka3, 2023-09-04:
Got some stuff working following your tips, thanks! Will be presenting this at upcoming VF2023 Austin Sep 14th. Looking forward to it!

Thanks

closed 2023-09-04 by svenka3