MikePopoloski/pyslang #208: visit default print

a-bian, 2023-11-02:
The pyslang now has a new 'visit' method, but it seems that replacement is not supported yet.

Is there a similar way to implement the following?
```
rep = {"CEN":"PEN", "DEN":"KEN"}
def visit_sub(obj):
    if obj.kind.name in ["NamedPortConnection"]:
        if str(obj.name) in rep:
            print(".%s (%s)," % (rep[str(obj.name)],obj.expr))
        else:
            print ... # default print
    else:
        print ...  # default print
```
However, when using the above approach, Python 3 crashes directly.

MikePopoloski, 2023-11-03:
Do you have a full reproducer for the crash? Otherwise I'm not sure exactly what you're asking for.

a-bian, 2023-11-03:
The issue has been resolved through other means, thank you!

closed 2023-11-03 by a-bian