MikePopoloski/pyslang #210: visit() will skip module instantiation in generate-for loop

riggy2013, 2023-11-03:
Below is the sample Verilog code:

```
genvar i;
generate
for(i=0; i<SIZE; i=i+1) begin:bitnum
  wire t1, t2, t3;
  xor g1 ( t1, a[i], b[i]);
  xor g2 ( sum[i], t1, c[i]);
  and g3 ( t2, a[i], b[i]);
  and g4 ( t3, t1, c[i]);
  or g5 ( c[i+1], t2, t3);

  mod_u i_u(
	  .i (a),
	  .o (d)
  );
end
endgenerate
```
My visit() code is as following:

```
def visit(self, node):

# below ln is the debug code where I am trying to see which line and node is visited.
    if hasattr(node, 'location'):
      ln = self.sm.getLineNumber(node.location)
    else:
      ln = 0
# debug code ends

    if isinstance(node, pyslang.PrimitiveInstanceSymbol) or isinstance(node, pyslang.UninstantiatedDefSymbol):
      ln = self.sm.getLineNumber(node.location)
      inst_name = node.name
#... my process with inst_name here

```

However it will process primitive instances but no mod_u i_u module.

MikePopoloski, 2023-11-03:
It's hard to tell without the full Verilog and Python code what the problem is. I need a full reproducer, since what you've shown me looks reasonable and isn't obviously wrong.

riggy2013, 2023-11-04:
Hi, Mike,

Pls see whether you can reproduce.

gen_inst.v:
```
module addergen1
#(parameter SIZE = 4)
(
  input  logic [SIZE-1:0] a, b,
  input  logic            ci,
  output logic            co,
  output logic [SIZE-1:0] sum
);

wire [SIZE :0] c;
wire i_u;
genvar i;

assign c[0] = ci;

// Hierarchical gate instance names are:
// xor gates: bitnum[0].g1 bitnum[1].g1 bitnum[2].g1 bitnum[3].g1
// bitnum[0].g2 bitnum[1].g2 bitnum[2].g2 bitnum[3].g2
// and gates: bitnum[0].g3 bitnum[1].g3 bitnum[2].g3 bitnum[3].g3
// bitnum[0].g4 bitnum[1].g4 bitnum[2].g4 bitnum[3].g4
// or gates: bitnum[0].g5 bitnum[1].g5 bitnum[2].g5 bitnum[3].g5
// Gate instances are connected with nets named:
// bitnum[0].t1 bitnum[1].t1 bitnum[2].t1 bitnum[3].t1
// bitnum[0].t2 bitnum[1].t2 bitnum[2].t2 bitnum[3].t2
// bitnum[0].t3 bitnum[1].t3 bitnum[2].t3 bitnum[3].t3

generate
for(i=0; i<SIZE; i=i+1) begin:bitnum
  wire t1, t2, t3;
  xor g1 ( t1, a[i], b[i]);
  xor g2 ( sum[i], t1, c[i]);
  and g3 ( t2, a[i], b[i]);
  and g4 ( t3, t1, c[i]);
  or g5 ( c[i+1], t2, t3);

  mod_u i_u(
	  .i (a),
	  .o (d)
  );
end
endgenerate

generate
	if (SIZE == 1) begin:u1
		mod_b u_gen_mod_a(
		.i(a),
		.o(b));
	end else begin:u2
		mod_c u_gen_mod_b(
		.i(a),
		.o(b));
	end
endgenerate

generate
	case (SIZE)
		1: begin
			mod_d u_gen_mod_d(
				.i(a),
				.o(b));
		end
		2: begin
			mod_e mod_e(
				.i(a),
				.o(b));
		end
	endcase
endgenerate

assign co = c[SIZE];

endmodule
```

my code, fix_vinst3.py:
```
#!/usr/bin/env python
# _*_ coding: utf-8 _*_

import pyslang
import argparse
import re

parser = argparse.ArgumentParser(description='Process some integers.')
parser.add_argument('filename', type=str, help='Input file name')
args = parser.parse_args()

def inst_rep(inst_name):
  if re.match(r'^u_\w+_\d+$', inst_name):
    return re.sub(r'^u_(\w+)_(\d+)', r'i\2_\1', inst_name)
  if re.match(r'^u_', inst_name):
    return re.sub(r'^u_', r'i_', inst_name)
  elif re.match(r'^u\d+_', inst_name):
    return re.sub(r'^u(\d+)_', r'i\1_', inst_name)
  elif re.match(r'\w+_u$', inst_name):
    return re.sub(r'(\w+)_u$', r'i_\1', inst_name)
  elif re.match(r'\w+_u\d+$', inst_name):
    return re.sub(r'(\w+)_u(\d+)$', r'i\2_\1', inst_name)
  elif re.match(r'\w+_inst$', inst_name):
    return re.sub(r'(\w+)_inst', r'i_\1',  inst_name)
  elif re.match(r'\w+_inst\d+$', inst_name):
    return re.sub(r'(\w+)_inst(\d+)$', r'i\2_\1', inst_name)
  else:
    return re.sub(r'^', r'i_', inst_name)

def check_members(sm, member):
  if not member:
    return

  if member.kind == pyslang.SyntaxKind.HierarchyInstantiation or \
    member.kind == pyslang.SyntaxKind.PrimitiveInstantiation:
    inst = member.instances[0]
    tok = inst.getFirstToken()
    ln = sm.getLineNumber(tok.location)
    if inst.decl != None:
      inst_name = inst.decl.name.value
    if not re.match(r'^i_|^i\d+_', inst_name):
      print('%d s/%s/%s/' % (ln, inst_name, inst_rep(inst_name.lower())))
  else:
    if hasattr(member, 'members'):
      for sub_memb in member.members:
        check_members(sm, sub_memb)
    if hasattr(member, 'items'):
      for sub_item in member.items:
        check_members(sm, sub_item)
    if hasattr(member, 'block'):
      check_members(sm, member.block)
    if hasattr(member, 'clause'):
      check_members(sm, member.clause)
    if hasattr(member, 'elseClause'):
      check_members(sm, member.elseClause)

def extract_instance_names(filename):
  tree = pyslang.SyntaxTree.fromFile(filename)
  sm = tree.getDefaultSourceManager()
  for mod in tree.root.members:
    check_members(sm, mod)

def extract_line(filename, ln):
  with open(filename, 'r') as file:
    lines = file.readlines()
  return lines[ln - 1]

def is_repeat_pattern(text, pat):
  return text.count(pat) > 1

class Visitor():
  def __init__(self, sm, filename):
    self.sm = sm
    self.filename = filename

  def visit(self, node):
    if hasattr(node, 'location'):
      ln = self.sm.getLineNumber(node.location)
    else:
      ln = 0
      
    if isinstance(node, pyslang.PrimitiveInstanceSymbol) or isinstance(node, pyslang.UninstantiatedDefSymbol):
      ln = self.sm.getLineNumber(node.location)
      inst_name = node.name
      if not re.match(r'^i_|^i\d+_', inst_name):
        rep_name = inst_rep(inst_name.lower())
        if is_repeat_pattern(extract_line(self.filename, ln), inst_name):
          print('# ', end='')
        print('%d s/%s/%s/' % (ln, inst_name, rep_name))

c = pyslang.Compilation()
tree = pyslang.SyntaxTree.fromFile(args.filename)
sm = tree.getDefaultSourceManager()
c.addSyntaxTree(tree)
v = Visitor(sm, args.filename)
c.getRoot().visit(v.visit)

#extract_instance_names(args.filename)
```

You can ignore the func of extract_instance_names/check_members which are my previous implementation. Please just run ./fix_vinst3.py gen_inst.v

MikePopoloski, 2023-11-04:
It appears your code is not processing the `mod_u i_u` instance because your code explicitly skips over instances that match this regex:
```
if not re.match(r'^i_|^i\d+_', inst_name):
```

closed 2023-11-09 by MikePopoloski