MikePopoloski/pyslang #220: Request for assistance

17Reset, 2023-11-28:
Can you tell me how the following example can be called correctly in pyslang to achieve the same effect?

slang top_module.sv -v some/lib1.sv -v "my_lib=some/other/lib2.sv"

MikePopoloski, 2023-12-02:
I added an example in 44b1172c9c010439c104fef429ba22137da0e133

17Reset, 2023-12-06:
I would like to know how to call pyslang in Python to generate a report, my previous code is shown below, how can I modify this to get the same result as above?
def sv_check(code_contents):
    tree = pyslang.SyntaxTree.fromText(code_contents)
    comp = pyslang.Compilation()
    comp.addSyntaxTree(tree)
    diags = comp.getAllDiagnostics()
    report = pyslang.DiagnosticEngine.reportAll(comp.sourceManager, diags)
    return report

MikePopoloski, 2023-12-06:
Is there something wrong with the code you posted? What is it doing that you want done differently?

17Reset, 2023-12-07:
The example I provided detects the content of a single file and reports an error if the detected file has an import of a resource file, but slang's command line mode has commands --libfile and --libdir to add a resource file or directory, and I'd like to use these two functions in my python code as well so that when I check a file, I don't get unrecognized imports when I specify a resource file or a directory so that when I detect a file, the result of the detection will not have unrecognized imports.

MikePopoloski, 2023-12-07:
Ok, that's what the example I added shows how to do -- use the Driver class to get all of the command line argument handling that the main slang tool provides. You can look at what the code in the Driver methods are doing and perform the steps yourself if you want to customize something.

17Reset, 2023-12-08:
Can you give me an example of my requirement how to use the Drive you mentioned above?Thank you very much.

17Reset, 2024-01-04:
source code：
```
def xqt_sv_check(file_list, slang_args=None):
    if slang_args is None:
        slang_args = []

    # Create a slang driver with default command line arguments
    driver = Driver()
    driver.addStandardArgs()

    # Parse command line arguments
    sv_check_args = ' '.join(SV_CHECK_ARG)
    slang_args = ' '.join(slang_args)
    file_list_args = ' '.join(file_list)
    fullcmd = f'{SV_CHECK_CMD} {sv_check_args} {slang_args} {file_list_args}'

    if not driver.parseCommandLine(fullcmd, CommandLineOptions()):
        return
    # Process options and parse all provided sources
    if not driver.processOptions() or not driver.parseAllSources():
        return

    # Perform elaboration and report all diagnostics
    compilation = driver.createCompilation()
    diags = compilation.getAllDiagnostics()
    report = DiagnosticEngine.reportAll(sourceManager=compilation.sourceManager, diag=diags)

    return report

print(xqt_sv_check(['D:/Download/SV_file/Project/project_a/rtl/code/async_fifo.v']))
```
error: Traceback (most recent call last):
  File "..\xqt_sv\xqt_sv_check.py", line 54, in <module>
    print(xqt_sv_check(['D:/Download/SV_file/Project/project_a/rtl/code/async_fifo.v']))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "..\xqt_sv\xqt_sv_check.py", line 36, in xqt_sv_check
    if not driver.processOptions() or not driver.parseAllSources():
           ^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: invalid utf8