MikePopoloski/pyslang #222: DPI pure task - compatibility relaxation request

svenka3, 2023-11-30:
Consider the below code:

```verilog
// BAD code - pure task is not allowed
// but some tools compile
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
// OK import "DPI-C"  task bad_pure_dpi_t (logic [3:0] a);

module top ;
endmodule

```

Below slang fails to compile:

```sh
slang --compat vcs ../sv_tests/test_issue_57_1_f.sv
```

Can you please relax this and make that available via PySlang (Am adding a rule in PySlint to check such issues).

Thanks

MikePopoloski, 2023-12-16:
Done in 267adb42233ad0db21edb00e6205c124a03a37d8

closed 2023-12-16 by MikePopoloski

svenka3, 2023-12-18:
Thanks for the fix. I rebuilt slang locally and I can compile this code with:
```sh
 slang ../sv_tests/test_issue_222_f.sv -Wdpi-pure-task
```
However, my pyslang still seems not to like it. I did:
```sh
 pip3 install -U pyslang
```
But still I see old behavior when PySlint parses the above code. Do I need to add this "-Wdpi-pure-task" somehow to my pyslang invocation options?

Thanks

MikePopoloski, 2023-12-18:
You need to wait for the next official release or build and install the pyslang package locally.

svenka3, 2023-12-30:
Quick follow-up: 

 pip3 install --upgrade pyslang
Requirement already satisfied: pyslang in /usr/local/lib/python3.8/dist-packages (5.0.0)

I did the above. I believe this fix is in 5.0.0 release (Correct me if I am wrong).  But pyslang still does not seem to parse this test case (As in this issue) - do I need to add "-Wdpi-pure-task" to options? If so, is it done via pyslang.Bag()? Please share an example of how to pass extra options in general to slang when using pyslang wrapper.

Thanks, happy new year!

MikePopoloski, 2023-12-31:
It depends how you're using the library. If you're using the Driver class then you can pass command line options to it directly. If you're creating a DiagnosticEngine you can configure it using the available methods, etc.

Happy new year to you as well.

svenka3, 2023-12-31:
Thanks. Below is what I do as of now:
```python
tree = pyslang.SyntaxTree.fromFile(inp_test_name)
r = tree.root
```
I did spend sometime looking at Driver.cpp and attempted the following:

```python
af_sv_bag = pyslang.Driver().createOptionBag()
pyslang.Driver().addStandardArgs()
 
af_sv_drv = pyslang.Driver

af_sv_copts = af_sv_bag.compilationOptions
```
I see Driver::addCompilationOption method is private, so perhaps no PY access? In the above af_sv_copts, I can't get to "flags" unlike the example/test in CPP, it says "no attribute named flags".

Thanks a lot for your time and efforts!

MikePopoloski, 2024-01-05:
If you use the Driver then you want to pass it your command line arguments via parseCommandLine(). createOptionBag calls addCompilationOptions so you don't need to call it yourself. Something like:
```
driver = pyslang.Driver()
driver.addStandardArgs()
driver.parseCommandLine("<command line here>")
driver.processOptions()
options = driver.createOptionBag()
```

If you already have code using just the SyntaxTree methods directly I'm not sure what the problem is. You can filter diagnostics you don't like out of the tree's diagnostics set, or if you're passing them to a DiagnosticEngine for rendering you can set options there to control what the output looks like.

svenka3, 2024-01-08:
Will open a new issue to seek help, thanks