MikePopoloski/pyslang #244: DPI pure task - adding extra option while using SyntaxTree

svenka3, 2024-01-08:
Following up on my recent request for DPI pure task, below is a complete code to demonstrate my issue.

```python
import pyslang
af_sv_bag = pyslang.Driver().createOptionBag()
pyslang.Driver().addStandardArgs()

af_sv_drv = pyslang.Driver
af_sv_drv = pyslang.Driver()
af_sv_drv.addStandardArgs()
af_sv_drv.parseCommandLine("-Wdpi-pure-task")
af_sv_drv.processOptions()
af_sv_bag = af_sv_drv.createOptionBag()

af_src_mgr = pyslang.SourceManager()

tree = pyslang.SyntaxTree.fromFile('dpi.sv', sourceManager=af_src_mgr, options=af_sv_bag)

print (tree.root.members)
```

With a test code:
```verilog

// BAD code - pure task is not allowed
// but some tools compile
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
// OK import "DPI-C"  task bad_pure_dpi_t (logic [3:0] a);

module top ;
endmodule
```

I am getting error as below:

```log
error: no input files

// BAD code - pure task is not allowed
// but some tools compile
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
```

I am sure I am missing something while using Bag, options and SyntaxTree. Can you please help me fix?

Thanks

MikePopoloski, 2024-01-10:
This seems a little confused. Every time you call `pyslang.Driver()` you're creating a new instance of the Driver class. In a lot of your code you seem to be immediately throwing away that instance? Create one instance and use that for all of the rest of the code.

Also, the Driver class assumes it's going to load all of your source files for you. If you're still going to manually load them with SyntaxTree.fromFile then you probably shouldn't bother with the Driver class. You can instead filter the diagnostics yourself when inspecting the SyntaxTree.diagnostics member, or by calling methods on a DiagnosticEngine if you're using that to print diagnostics.

svenka3, 2024-01-10:
That "Driver" call was run-away code left during copy-paste, sorry for the confusion. Yes, I don't use Driver() so far, instead am able to get things done with SyntaxTree.fromFile (Would love to learn more about the advantages of using Driver - as a separate topic). My real issue is how to pass additional option while using SyntaxTree.fromFile? Sort of similar to af_sv_drv.parseCommandLine("-Wdpi-pure-task") - but while loading files manually. 

Thanks

svenka3, 2024-01-11:
Apologies for not replying on diagnostics - that's simply because I am yet to use that. I do not report diagnostics yet in my code. I looked at driver.py example and coded the following, using driver:
```python
import sys

from pyslang import *

def main():
    # Create a slang driver with default command line arguments
    driver = Driver()
    driver.addStandardArgs()

    # Parse command line arguments
    args = " ".join(sys.argv)
    args += " -Wdpi-pure-task"
    #if not driver.parseCommandLine(args, CommandLineOptions()):
    if not driver.parseCommandLine(args):
        return

    # Process options and parse all provided sources
    if not driver.processOptions() or not driver.parseAllSources():
        return

    # Perform elaboration and report all diagnostics
    compilation = driver.createCompilation()
    driver.reportCompilation(compilation, False)
    print ((driver.syntaxTrees[0].root))

if __name__ == "__main__":
    main()
```

With same test case as before (pasted below for ease);
```verilog
// BAD code - pure task is not allowed
// but some tools compile
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
// OK import "DPI-C"  task bad_pure_dpi_t (logic [3:0] a);

module top ;
endmodule
```
I run it as:
```sh
python3 drv.py dpi.sv
```
I get a warning as:
```log
warning: unknown warning option '-Wdpi-pure-task' [-Wunknown-warning-option]
```

Does that mean I have old version of pyslang/slang still? 

Then I get errors as below:
```log
dpi.sv:4:21: error: expected 'function'
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
                    ^
dpi.sv:4:21: error: expected data type (implicit type name not allowed)
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
                    ^
dpi.sv:7:1: error: expected statement
module top ;
^~~~~~
dpi.sv:8:10: error: expected 'endtask'
endmodule
         ^
dpi.sv:4:22: note: to match this 'task'
 import "DPI-C" pure task bad_pure_dpi_t (logic [3:0] a);
                     ^
warning: no top-level modules found in design [-Wmissing-top]

Build failed: 4 errors, 2 warnings
```

Thanks

MikePopoloski, 2024-01-11:
Yes, your example works for me with the latest version of pyslang. You should upgrade your local copy (the current version is 5.0.0).

svenka3, 2024-01-11:
Thanks, issue was I had 2 installs of pyslang with .local having older version, now fixed. Is there a way to pass that additional arg while using syntaxTree.fromFile()? I will eventually move to Driver() usage as it seems more capable.

Thanks again for your wonderful support!

MikePopoloski, 2024-01-11:
SyntaxTree doesn't take options for controlling diagnostic output because it doesn't perform diagnostic output. You're free to filter the diagnostics produced by SyntaxTree before printing them yourself.

MikePopoloski, 2024-01-19:
Feel free to reopen if you have more questions.

closed 2024-01-19 by MikePopoloski