MikePopoloski/pyslang #251: Is there a way to create a single syntax tree from multiple sources

Nalaka1693, 2024-02-07:
I have two modules in two files
mux.v 
```
`define WIDE 3

module andd(output c, input a, input b);
    assign c = a & b;
endmodule
```
tok.v
```
module test(input sel, input [`WIDE:0] data0, input [`WIDE:0] data1, output reg out, output out0);
  always @(*) begin
    case (sel)
     ......
    endcase
  end

  andd a0 (out0, data0[0], data1[0]);
endmodule
```

if I create a driver and pass the files I only get one syntax tree with one module. is there a way to get both into a single tree?
Also, can I get the final tree honoring the macro definitions of the given file order? Thank you
```
driver = pyslang.Driver()
driver.addStandardArgs()
cmdOpts = pyslang.CommandLineOptions()

cmd = "mux.v tok.v"
driver.parseCommandLine(cmd, cmdOpts)

driver.processOptions()
driver.parseAllSources()

c = driver.createCompilation()

c.getSyntaxTrees()
[<pyslang.SyntaxTree ...fd7f46670>]

list(c.getSyntaxTrees()[0].root.members)
[SyntaxNode(SyntaxKin...claration)]

c.getDefinitions()
[DefinitionSymbol("mux")]
```

MikePopoloski, 2024-02-08:
It sounds like you want to pass the `--single-unit` flag. By default all files are put into separate compilation units.

Nalaka1693, 2024-02-08:
Thank you it worked I had to change the command to below, exactly in this order.
```
cmd = "-v mux.v tok.v --single-unit"
```

Below only gave the mux.v content as a tree.
```
cmd = "tok.v -v mux.v tok.v --single-unit"
```

closed 2024-02-09 by Nalaka1693