MikePopoloski/pyslang #254: Need help - get handle to the class from a method prototype

svenka3, 2024-02-12:
Hi,
  I am trying to get to the class that contained a method (task t in example below). I guess SyntaxTree won't have this information really, is there another way to get to it (Ast?).

Thanks
```verilog
class c;
  rand logic l1;
  function void f1();
  endfunction
  extern task t();
endclass : c
task c::t();
  $display ("Hello");
endtask : t
```

Pyslang code:
```python
import pyslang
tree = pyslang.SyntaxTree.fromFile('test.sv')

for s in (tree.root.members):
  if (s.kind.name == 'TaskDeclaration'):
    print ((s.prototype))
    #Q - how do I get to class c and iterate over its members again here? 
```

MikePopoloski, 2024-02-13:
Yes, you need to either use the AST or reimplement the lookup mechanics that the AST uses yourself.

MikePopoloski, 2024-03-05:
Closing until there's more to do here.

closed 2024-03-05 by MikePopoloski