MikePopoloski/pyslang #260: segfault crash when visiting syntax containing empty list

anders-code, 2024-02-25:
I ran across this segfault. Seems like when visiting certain nodes, they are corrupt (probably uninitialized memory). If you skip them, nothing bad happens. For example, if you just look for type() == Token and only print those, everything is OK. But touching anything except type() or kind on some nodes will crash (segfault) the python interpreter.

Curiously, you can visit the same tree using python and the bad nodes are OK. Looks like this is happening on empty len() == 0 SyntaxList nodes.

```
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyslang
>>> pyslang.__version__
'5.0.10'
```

```python
#!/usr/bin/python3

import pyslang

# this makes the first syntax list non-zero length and doesn't crash until 2nd
#t = pyslang.SyntaxTree.fromText("(* KEEP = \"true\" *) wire [31:0]a;");

t = pyslang.SyntaxTree.fromText("wire [31:0]a;");

def visit_ok(node):
    print(f"{type(node)} {node.kind} ({node.kind.value}): ", end="")

    if type(node) == pyslang.SyntaxNode:
        # i think this is a SyntaxList, but it's corrupted
        # if you touch anything in it, segfault
        print("XXXX skipping!")
        return

    print(node)

def visit_crash(node):
    print(f"{type(node)} {node.kind} ({node.kind.value}): ", end="")
    print(node)

def myvisit(node, v):
    v(node)
    if hasattr(node, "__iter__"):
        for c in node:
            myvisit(c, v)

print("\n----------------")
print("using visit with workaround\n")
t.root.visit(visit_ok)

print("\n----------------")
print("visiting manually without workaround\n")
myvisit(t.root, visit_crash)

print("\n----------------")
print("using visit without workaround - crash\n")
t.root.visit(visit_crash)
```

MikePopoloski, 2024-03-05:
Thanks for the detailed report. I fixed the bug upstream in e616f01ced49e898bbd8f88fe0050ce63db62088

closed 2024-03-05 by MikePopoloski

anders-code, 2024-03-06:
No problem!

I have a couple more small test cases that I have found. Is an issue in this project in a similar format the best way to report them? Is it better to make one issue or a couple?

I think one is a crash and another is a python exception or something like that. I should update to the fixed version and retest, but I think they are unique and will still fail.

MikePopoloski, 2024-03-06:
Sure, multiple issues is fine.