MikePopoloski/pyslang #312: Expose API to IDEs through .pyi files

adwait, 2024-06-04:
(feature proposal)

Some editors (e.g., vscode with pylance) do not load *.so files for Python API hints (https://github.com/microsoft/pylance-release/issues/242#issuecomment-674288520). 

In such cases it would good to have .pyi stub files. This can be autogenerated using tools such as mypy/stubgen (https://mypy.readthedocs.io/en/stable/stubgen.html). However, this will require some work on the cython library packaging.

The pydrake bindings for the drake C++ library did this sucessfully: https://github.com/RobotLocomotion/drake/issues/16987

adwait, 2024-06-05:
Suggested previously here: https://github.com/MikePopoloski/pyslang/issues/96

However, this does not work for me out of the box on version 6.0.

MikePopoloski, 2024-06-06:
I agree that this is nice to have but it's not something I have time to spend on any time soon. I'd be happy to accept a PR though if you want to tackle it.

parker-research, 2024-12-17:
I'd be happy to tackle this!

Both of the following generation methods work:
```bash
pip install mypy
stubgen -p pyslang

pip install pybind11-stubgen
pybind11-stubgen pyslang
# Note: Lots of error log messages. Generation still succeeds though.
```

Curious if anyone has a preference about which one (mypy's `stubgen` or `pybind11-stubgen`) is better? Seems like the pybind11 one probably generates richer stubs.

Where in the build process should the stub files be generated? Should they be generated on every pushed commit and version-controlled? Or should they only be generated during the release pipeline? I'm not too sure what the standard is. I think the standard is probably version-controlled.