MikePopoloski/pyslang #354: Unknown error

17Reset, 2024-10-08:
I used python code like the one below, with an error at pyslang==7.0, which worked perfectly at pyslang==6.0, but couldn't get any error messages. If the main process calls the following function, there will be no error, if it is placed in the child process, there will be an error
```
def rtl_check(
    file_list: list[str], slang_args: list[str] | None = None
) -> str | None:
    # Initialize the slang driver and add standard command-line arguments.
    driver = Driver()
    driver.addStandardArgs()

    # Construct the full command to be passed to the slang driver.
    full_cmd = [SV_CHECK_CMD] + SV_CHECK_ARG + (slang_args or []) + file_list
    full_cmd = ' '.join(full_cmd)

    # Parse the command-line arguments using the slang driver.
    if not driver.parseCommandLine(full_cmd, CommandLineOptions()):
        return None

    # Process options and parse all provided source files.
    if not driver.processOptions() or not driver.parseAllSources():
        return None

    # Create a compilation object and retrieve all diagnostics.
    compilation = driver.createCompilation()
    diags = compilation.getAllDiagnostics()
    source_manager = compilation.sourceManager

    # Generate and return the diagnostic report.
    return DiagnosticEngine.reportAll(source_manager, diags)
```

17Reset, 2024-10-08:
I found through printing that in pyslang==6.0 `driver.parseAllSources()` will return a True, but in Pyslang 7.0, `driver.parseAllSources()` directly error, no error message

MikePopoloski, 2024-10-11:
There is not enough information here to diagnose the problem. What is the main process and child process? What do your inputs look like? Are you capturing stderr?

17Reset, 2024-10-12:
If I run the above python function in a python script, `if __name == '__main__'`, the script does not get the error and gets the detection message normally. However, I am in PyQt6, using the multiprocess module to start a child process to execute the above function, and then the above uncatchable error occurs. Mainly the program is using pyslang 6.0 without any error.

17Reset, 2024-10-15:
OS: Windows11 Pro 23H2
Python: 3.12.7

Library Install:
```
pip install pyslang==7.0
pip install PyQt6
pip install psutil
```

Run:
```
D:\Job\Programming\python\xTest\xPyQt6\pyslang(sv check)>python sv_check_gui.py
```

Description:
I made a simple copy of the code, as shown below. I use Pycharm for development, when I run this file in Pycharm, there is no error, when running this file in Terminal through the script, pyslang7.0 will cause crash, I have found the crash site in 'driver.parseAllSources()', but there is no error message, so I can't go any further. The following code works perfectly at pyslang==6.0.

sv_check_gui.py
```python
import random
import shutil
import string
import sys
import os
import tempfile
from pathlib import Path
from multiprocessing import Process, Queue
import psutil
from PyQt6.QtWidgets import (
    QApplication, QMainWindow, QWidget, QVBoxLayout, QPushButton, QTreeWidget,
    QTreeWidgetItem, QFileDialog, QTextEdit, QSplitter
)
from PyQt6.QtCore import Qt, QThreadPool, QTimer, QRunnable, pyqtSignal, QObject
from pyslang import Driver, CommandLineOptions, DiagnosticEngine

PROCESS_TERMINATED = 'PROCESS_TERMINATED'
PROCESS_STOP = 'PROCESS_STOP'

SV_CHECK_CMD: str = 'slang'
SV_CHECK_ARG: list = ['--timescale 1ns/1ps']

def sv_check(
    file_list: list[str], slang_args: list[str] | None = None
) -> str | None:
    print(file_list)

    # Initialize the slang driver and add standard command-line arguments.
    driver = Driver()
    driver.addStandardArgs()

    # Construct the full command to be passed to the slang driver.
    full_cmd = [SV_CHECK_CMD] + SV_CHECK_ARG + (slang_args or []) + file_list
    full_cmd = ' '.join(full_cmd)

    # Parse the command-line arguments using the slang driver.
    if not driver.parseCommandLine(full_cmd, CommandLineOptions()):
        return None

    # Process options and parse all provided source files.
    if not driver.processOptions() or not driver.parseAllSources():
        return None

    # Create a compilation object and retrieve all diagnostics.
    compilation = driver.createCompilation()
    diags = compilation.getAllDiagnostics()
    source_manager = compilation.sourceManager

    # Generate and return the diagnostic report.
    return DiagnosticEngine.reportAll(source_manager, diags)
[... 250 lines cut ...]
        data_queue = Queue()
        file_list = get_file_list(self.tmp_dir, ['*.v', '*.sv'])
        data_queue.put(file_list)

    def code_check_display(self, report) -> None:
        if report:
            self.check_canvas.setPlainText(report)

    def closeEvent(self, a0):
        if self.check_process:
            self.check_process.stop()
        if self.thread_pool:
            self.thread_pool.clear()

        shutil.rmtree(self.tmp_dir)

if __name__ == "__main__":
    app = QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec())
```

Verilog files
[check_file.zip](https://github.com/user-attachments/files/17374419/check_file.zip)

MikePopoloski, 2024-10-20:
I was able to reproduce this. I believe it is this issue: https://github.com/actions/runner-images/issues/10004#issuecomment-2156109231

Basically PytQt6 ships with its own version of the MSVC redist that is too old. It happens to get loaded first in the subprocess and therefore causes the slang module to crash when trying to use a mutex. You can work around it for now by telling slang not to use multiple threads with a `j1` command line argument. I can also patch the slang Python build to include the `_DISABLE_CONSTEXPR_MUTEX_CONSTRUCTOR` define which should help backwards compatibility.

17Reset, 2024-10-21:
Thanks, Expect new wheels to be built，including python3.13

MikePopoloski, 2024-11-04:
Fixed in 7.0.1