MikePopoloski/pyslang #380: Missing basic AST equality functionality

parker-research, 2024-11-28:
The following test seems like it should pass, but it does not:

```python
def test_verilog_equality() -> None:
    """Test that two Verilog modules with minor differences are evaluated as equal."""

    input_1 = """
        module and_gate (
            input wire x,
            input wire y,
            output wire z
        );
            assign z = x & y;
        endmodule
    """

    st1 = pyslang.SyntaxTree.fromText(input_1)
    st1_again = pyslang.SyntaxTree.fromText(input_1)
    st1_stripped = pyslang.SyntaxTree.fromText(input_1.strip())

    assert st1 == st1_again # fails
    assert st1 == st1_stripped # fails

```

MikePopoloski, 2024-11-28:
Why would you expect syntax nodes from different files to compare equal? That would be surprising to me, since they come from separate files and therefore have separate source locations. If you want to convert the syntax nodes to text format and compare the strings it's not hard to do so.

parker-research, 2024-11-30:
Alright, fair enough! I agree, given this insight!

Can you recommend what text representation is best to compare, with things like whitespaces not affecting the comparison? I feel like this sort of transformation of code, ignoring semantics like whitespaces and potentially comments, is a very common use case of these types of AST libraries. Might be a good example to add!

MikePopoloski, 2024-11-30:
There is an `isEquivalentTo` function on SyntaxNodes that compare logical equivalence (i.e. without any trivia like whitespace or comments), which is maybe what you want. Otherwise if you want a string comparison you can use the SyntaxPrinter class which has various options for controlling whether comments, trivia, preprocessed nodes, etc are included.

parker-research, 2024-12-01:
I suppose the final point on this then is just that it'd be nice if these features were documented better!

MikePopoloski, 2024-12-02:
Indeed, as discussed in other threads, always better to have more in-depth documentation.

parker-research, 2025-01-13:
Why doesn't `SyntaxTree` have an `isEquivalentTo` method?

MikePopoloski, 2025-01-13:
Because nobody ever needed it. All features are unimplemented by default.