MikePopoloski/pyslang #381: Request: Add docs on how these bindings are created, how to contribute, etc.

parker-research, 2024-11-28:
Related to the discussion in #65 - I'd like to help create the bindings discussed there.

Do you have any info on how the bindings are generated, or a guide, or anything?

This repo is very bare as far as docs go. A basic "how to contribute" guide with a link to the examples you used to learn how to make this repo in the first place would be very helpful.

MikePopoloski, 2024-11-28:
This library uses pybind11 to generate the bindings, which is pretty much the industry standard for exposing C++ APIs to Python. Reading through the pybind11 docs should tell you all you need to know. There isn't really anything else; the bindings themselves are in the upstream project and this repo just contains the Python packaging script and some simple tests.

parker-research, 2024-11-30:
Even that much information, verbatim in a `CONTRIBUTING.md` file would be helpful!

I can potentially submit that if I get a change!

MikePopoloski, 2024-11-30:
Yeah, there's a CONTRIBUTING file in the upstream repo: https://github.com/MikePopoloski/slang/blob/master/CONTRIBUTING.md

I suppose it can't hurt to have one here that points to that one. This is really just a packaging repo.