MikePopoloski/pyslang #382: Request: Add Python type hinting to pyslang

parker-research, 2024-11-28:
Pyslang does not currently appear to be configured/set up to support strong typing at all, making it a bit difficult to use with modern IDEs (which can help suggest whether a method actually exists, etc.). 

It would drastically increase the usability of this library if the typing information (which is presumably strong in C++'s type system) was transferred to pyslang.

MikePopoloski, 2024-11-28:
I agree that this is nice to have but it's not something I have time to spend on any time soon. I'd be happy to accept a PR though if you want to tackle it.

parker-research, 2025-01-13:
I figured out how to build the type stubs, but can't seem to figure out how to get wheel/build/cmake/setuptools/scikit-build to package it inside the wheel zip.

Going to have to take a break. Would really appreciate if you could take a look, as you might be a bit more familiar with how this all gets packaged up.

To generate the stubs, you can do:

```bash
pip install . # Install pyslang
pip install pybind11-stubgen
pybind11-stubgen pyslang -o stubs/
ls stubs/ # See the pyi file in it.
```

Now, just need to figure out how to ensure that `pyslang.pyi` file makes its way into the wheel file.

parker-research, 2025-01-14:
I've fully solved this and have the generation working rather well.

It seems like there are a few ways to automatically update the type stubs file:

1. We could require that it's manually generated, and have a CI check that ensures it's been updated.
2. We could add its generation to the `dist.yml` workflow.
3. We could add its generation to a GitHub Action that works similar to precommit (runs whenever there's a PR, and then tacks on a commit if necessary).
4. We could add it to the real pre-commit flow. However, because the process involves a complete build of the C code, it would be very slow, so this option might not be the best.

It seems that some repos like to check this file into version control as they end up making changes to it (probably doesn't apply here, at least at the start), and it seems some repos like to generate it at build/release-time.

Please let me know what you think is best, and I'll do my best to make it happen!

MikePopoloski, 2025-01-15:
Hmm, I think adding it to the dist workflow is the lightest touch right now so let's do that. If in the future we need to start checking it in we can look at something like option 3.