MikePopoloski/pyslang #385: CI pipeline doesn't seem to be running pytest right now

parker-research, 2024-12-01:
There is currently a failing unit test in `tests/test_basics.py`.

It seems that the pytest unit tests aren't being run in this repo's CI workflow right now. It would be great if it did! 

```
________________________________________________________ test_scriptsession ________________________________________________________

    def test_scriptsession():
        session = ScriptSession()
        session.eval('integer arr[string] = \'{"Hello":4, "World":8, default:-1};')
        session.eval(
            """
    function int func(int i, integer arr[string]);
        case (i) inside
            [32'sd1:32'sd2]: return 1;
            arr: return 2;
        endcase
        return 3;
    endfunction
    """
        )
    
        assert session.eval("func(4, arr)").value == 2
>       assert len(session.getDiagnostics()) == 0
E       assert 1 == 0
E        +  where 1 = len(<pyslang.Diagnostics object at 0x71edac84f5f0>)
E        +    where <pyslang.Diagnostics object at 0x71edac84f5f0> = getDiagnostics()
E        +      where getDiagnostics = <pyslang.ScriptSession object at 0x71edac84e870>.getDiagnostics

tests/test_basics.py:102: AssertionError
===================================================== short test summary info ======================================================
FAILED tests/test_basics.py::test_scriptsession - assert 1 == 0
```

MikePopoloski, 2024-12-02:
Hmm, they are supposed to be. I will look to see why they're not.

MikePopoloski, 2024-12-02:
No, they definitely still are running. For example from the most recent dependabot merge: https://github.com/MikePopoloski/pyslang/actions/runs/12077955465

parker-research, 2024-12-02:
It's quirky that this button doesn't show that. I guess the actions run on the real commit and not on the "merged pull request" commit (which ends up showing as the latest commit)

[image]

Regardless, it's too bad the tests are failing. I can't figure out what the appropriate way to fix them is.

MikePopoloski, 2024-12-02:
It's just a recently added warning in the upstream repo that fires on the test code. It's simple enough to filter it out in the test or to just count it as an expected result instead of asserting that there are no warnings issued.

MikePopoloski, 2024-12-02:
Just adding a `default` clause to the case would fix it as well.

PR #388 (2024-12-03): Fix ScriptSession diagnostic test failing
Fixes #385, indirectly
closed 2024-12-03 by MikePopoloski