MikePopoloski/pyslang #406: Docs: `isEquivalentTo` returns True when it should not

parker-research, 2025-01-14:
Consider the following test case, which I believe should fail, but which currently passes.

```python
import pyslang

def is_verilog_equal(verilog_left: str, verilog_right: str) -> bool:
    """Check if two Verilog snippets are equivalent."""
    ast_left = pyslang.SyntaxTree.fromText(verilog_left)
    ast_right = pyslang.SyntaxTree.fromText(verilog_right)
    return (ast_left.root).isEquivalentTo(ast_right.root)

def test_is_verilog_equal_when_not_equal() -> None:
    """Assert `is_verilog_equal` when the two Verilog modules are not equal."""
    input_1 = """
        module and_gate (
            input wire x,
            input wire y,
            output wire z
        );
            assign z = x & y;
        endmodule
    """

    input_2 = """
        module or_gate (
            input wire x,
            input wire y,
            output wire z
        );
            assign z = x | y;
        endmodule
    """

    assert is_verilog_equal(input_1, input_2) is False
```

MikePopoloski, 2025-01-15:
Indeed, this looks like a bug.

parker-research, 2025-01-15:
Is it supposed to compare that node **and descendents**, or just that node? I suppose that's the most important part to document first